khoih-prog/ESP32TimerInterrupt #1: ISR doesn't run unless interval is under 5000 microseconds.

IVData, 2020-09-03:
If I call `attachInterruptInterval` using `ESP32Timer myTimer(1);` with an interval above 5000 microseconds, the ISR never runs.

Any ideas?

khoih-prog, 2020-09-03:
Hi @IVData 

You have to start with the example to be familiar with the usage. For example [TimerInterruptTest](https://github.com/khoih-prog/ESP32TimerInterrupt/tree/master/examples/TimerInterruptTest).

The timers in that  [TimerInterruptTest](https://github.com/khoih-prog/ESP32TimerInterrupt/tree/master/examples/TimerInterruptTest) example are very long, 15,000 ms = 15,000,000 microseconds !!!

```
#define TIMER0_INTERVAL_MS        1000
#define TIMER0_DURATION_MS        5000

#define TIMER1_INTERVAL_MS        3000
#define TIMER1_DURATION_MS        15000
```

Basically, this library is designed for millisecond ranges, not for ranges under microseconds.

If you still have issue, **post your code** so that we can **duplicate** and see what's going wrong.

closed 2020-09-04 by khoih-prog

IVData, 2020-09-04:
Sorry for the poor bug report, it was getting late and I was desperate =P

I made a minimal example and in that sketch the library behaves correctly so it must be something on my end.

khoih-prog, 2020-09-04:
That's good to know. You definitely need some coffee and rest.

If there anything more, don't hesitate to post an issue, just when you have fresh mind and after desperation has gone ;-)

IVData, 2020-09-06:
Just in case in helps someone else, I figured out what was causing my issue.

I had an external source (SBUS signal) updating the interval for my timer. The issue is that running `updateInterval` resets the timer and starts the interval again. So if you update the interval at a faster rate than the interval itself, your ISR function will never run :)

I have solved by only updating my interval in the ISR itself.