khoih-prog/ESP32TimerInterrupt #5: a quick question

m-menon, 2020-09-18:
hi, I am about to try something with this timer before that I thought I'd have a word with you; Do you think it will be fine to stop or restart the timer from within the timer callback? Just checking if there could be any known issues with that? And thanks for the excellent library. I have just started using it.

khoih-prog, 2020-09-19:
@m-menon 

Thanks for using the library and encouraging words.

It's difficult to know what you'd like to know and to achieve, I suggest to go through all the examples to understand all the complicated aspects of the Timer Interrupt, and use it correctly and effectively.

You can see you can stop/start the ISR timers as many times as you like, in the [**TimerInterruptTest example**](https://github.com/khoih-prog/ESP32TimerInterrupt/tree/master/examples/TimerInterruptTest).

Also have a look at some more complicated examples in [**SmallProjects**](https://github.com/khoih-prog/SmallProjects), for example [**FireSmokeAlarm**](https://github.com/khoih-prog/SmallProjects/tree/master/FireSmokeAlarm)

Have a good time exploring,

m-menon, 2020-09-19:
@khoih-prog thanks for the links and the tips. I will explore the projects and see how it all goes. I'll close this now. Cheers.

closed 2020-09-19 by m-menon