khoih-prog/ESP32TimerInterrupt #8: RPM_Measure example does not work

Robbi-Blechdose, 2020-11-29:
This library looks perfect for my new project and as such I'd like to use it.
However, the RPM_Measure example (which is what I'll need in my project) does not work.
I've connected a reed switch with magnets on a disk in front of it to my ESP32, and the sketch prints bogus values with no discernible pattern. Except for the fact that all values are very high, usually between 500 and 740 RPM.
Is this a known issue or is there perhaps a working test setup?

khoih-prog, 2020-11-29:
Thanks for your interest in this [**ESP32TimerInterrupt Library**](https://github.com/khoih-prog/ESP32TimerInterrupt).

As you experience, it's very difficult to use and apply the code in the real world, unless the hardware, the settings, the code must be good, well understood, and designed / modified for the use case. Many experiments also must be done to achieve to correct results.

The principle of operation is quite simple

```
RPM = 60000 / (rotation time in ms)
```

If you read the notes inside the example

```
RPM Measuring uses high frequency hardware timer 1KHz == 1ms) to measure the time from of one rotation, in ms
then convert to RPM. One rotation is detected by reading the state of a magnetic REED SW or IR LED Sensor
Asssuming LOW is active.
For example: Max speed is 600RPM => 10 RPS => minimum 100ms a rotation. We'll use 80ms for debouncing
If the time between active state is less than 8ms => consider noise.
RPM = 60000 / (rotation time in ms)
```

The example is just an illustration for low-speed RPM measurement, max 600RPM, and the 80ms debouncing time is used in the example.

That's why the min. rotationTime must be 81+ (**because you always make it active == LOW at the PIN_X**)

You have to calculate and modify the example to fit your use case.

For example, if your max speed is 3000RPM = 20ms rotationTime. Use the debouncing time of 15-17 (must be **just less then min. rotationTime**).

Also be sure that **LOW is active**, which means that 

1. If using REED SW, it must be 
  - NO (normally open), with pull-up resistor to be sure, if connected between PIN_X and GND, **Advisable**.
  - NC (normally closed), with pull-down resistor, if connected between PIN_X and +3.3V.
2. If using LED, follow the same principle.

If you experience a very high speed (almost max. RPM or min rotationTime) such as you're experiencing, I think you connect the REED SW not correctly (LOW all the time, HIGH when active).

Also be sure  the REED SW, when active (for example NO, active => SW closed), the closing time must be longer than the debouncing time, or you won't detect any speed.

So, please spend some more time to really understand **how the measurement works, then modify the settings according to your use case (max RPM, etc), connect the REED SW correctly, etc.**

Good Luck and please post your experience back here to help other people if you're successful.

closed 2020-11-29 by khoih-prog

khoih-prog, 2020-11-30:
Another note if you have to use some kind of REED SW and connection is  **active == HIGH at the PIN_X**, just change the code as follows:

```
void IRAM_ATTR TimerHandler0()
{
  static bool started = false;

  if (!started)
  {
    started = true;
  }

  if ( digitalRead(SWPin) && (debounceCounter >= DEBOUNCING_INTERVAL_MS / TIMER0_INTERVAL_MS ) )
  {
    //min time between pulses has passed
    RPM = (float) ( 60000.0f / ( rotationTime * TIMER0_INTERVAL_MS ) );

    avgRPM = ( 2 * avgRPM + RPM) / 3,

    Serial.println("RPM = " + String(avgRPM) + ", rotationTime ms = " + String(rotationTime * TIMER0_INTERVAL_MS) );

    rotationTime = 0;
    debounceCounter = 0;
  }
  else
  {
    debounceCounter++;
  }

  if (rotationTime >= 5000)
  {
    // If idle, set RPM to 0, don't increase rotationTime
    RPM = 0;
    Serial.println("RPM = " + String(RPM) + ", rotationTime = " + String(rotationTime) );
    rotationTime = 0;
  }
  else
  {
    rotationTime++;
  }
}
```

The REED SW open/closing time (to make active HIGH at PIN_X) must be longer than your selected debouncing time.

Robbi-Blechdose, 2020-12-23:
As it turns out, this is indeed a problem with the example - the code doesn't check for edges on the signal.
Instead, it simply assumes the signal will be back to HIGH after the "debounce time" (which is not what debouncing is for).
Which fails quite easily depending on the setup.
In my case, the magnets were rather large and made up about half of the rotation.
Detecting clock edges reliably solves this problem and works fine even at high RPMs.

khoih-prog, 2020-12-23:
Thanks to share your experience with other users.

As you already know, the examples are just for **demonstrating the basic usage of the library features/functions, and are not intended to be used directly into any use case.** 

Many people also complain the examples, for other libraries, are over-complicated by taking care of too many possible scenarios. 

There's no way to satisfy everybody, with experience and knowledge ranging from novice to expert.

You can have a look at other examples to demonstrate the usage of edge detection [**ISR_Timer_4_Switches**](https://github.com/khoih-prog/SmallProjects/blob/master/ISR_Timer_4_Switches/ISR_Timer_4_Switches.ino) and hopefully it's somehow helpful to you.

Good Luck,