khoih-prog/ESP32TimerInterrupt #11: Clarifications on how the library works and ESP32-S2 support

RoscoeTheDog, 2021-03-23:
I am wondering if the newer ESP32-S2 is supported under this library. The ESP32-S2 has slightly different hardware (i.e 1 core), and if that would restrict the number of running ISR's?

I've dabbled with Arduino hardware timers before and from what I understand, the ISR's always been limited by the number of available compare registers for each timer. If the ESP32 has only 2 hardware timers as mentioned in the `ESP32_ISR_Timer.cpp` source file, then how does this library achieve 16 ISR's?

I was looking through some of the src myself and this snippit caught my attention:

```
      // is it time to process this timer ?
      // see http://arduino.cc/forum/index.php/topic,124048.msg932592.html#msg932592

      if ((current_millis - timer[i].prev_millis) >= timer[i].delay) 
      {
        unsigned long skipTimes = (current_millis - timer[i].prev_millis) / timer[i].delay;
        
        // update time
        timer[i].prev_millis += timer[i].delay * skipTimes;

        // check if the timer callback has to be executed
        if (timer[i].enabled) 
        {

          // "run forever" timers must always be executed
          if (timer[i].maxNumRuns == RUN_FOREVER) 
          {
            timer[i].toBeCalled = DEFCALL_RUNONLY;
          }
          // other timers get executed the specified number of times
          else if (timer[i].numRuns < timer[i].maxNumRuns) 
          {
            timer[i].toBeCalled = DEFCALL_RUNONLY;
            timer[i].numRuns++;

            // after the last run, delete the timer
            if (timer[i].numRuns >= timer[i].maxNumRuns) 
            {
              timer[i].toBeCalled = DEFCALL_RUNANDDEL;
            }
```

It looks like the main thread is iterating through each timer, checking the delta time between the last checked time, then comparing that to the specified ISR run-time in order to determine when to run a routine.

Wouldn't this mean these are technically software-based timers? Also if the main thread were hung up or processing something else, then wouldn't it not execute the timers immediately since it actually takes additional CPU cycles to run the code that checks the time comparisons?

khoih-prog, 2021-03-23:
Hi @RoscoeTheDog 

Thanks for your interest in the [**ESP32TimerInterrupt** library](https://github.com/khoih-prog/ESP32TimerInterrupt).

**It hasn't supported the ESP32-S2 yet**. Possibly in the future if there are more requests.

---

If you check [**Features**](https://github.com/khoih-prog/ESP32TimerInterrupt#features) more carefully, you'll see this library has no miracle to provide you 16 independent ISRs (while ESP32 has only 2), but only **16 ISR-based Timers, while consuming only 1 Hardware Timer**.

> Wouldn't this mean these are technically software-based timers?

It's up to you to say what you believe in, but I suggest you take some more time to go thru the examples to fully understand. 

---

> I was previously designing a circuit around an ATMega2560 chip because there is 11 hardware based ISR's available. But if I can get 16 out of a single ESP32, then I'd reconsider my circuit altogether.

Certainly, if you rewrite your code carefully and take time to understand more about this library, you possibly can achieve the result you'd like.

I'm afraid I don't have time to spend more on this issue and I'm closing it now.

closed 2021-03-23 by khoih-prog

RoscoeTheDog, 2021-03-23:
I can confirm that this does **not** currently work with ESP32-S2 (at least, specifically with the Saola-1 board). I attribute this due to the Arduino framework's timers for this board only being half-way supported right now (functions like millis() and delay() work, but not other ISR/Timer-based code). Once another update for the platformIO-arduin-idf framework is released and fully supports timers then this library _might_ work in the future since it relies on the Arduino framework it seems.. Khoih-prog would have to follow up on this and confirm it since I am not as familiar with the source code.

Thanks for the reply and the good work regardless. I am having a lot of compatibility issues with this board and I might just return/exchange it for a normal esp32 instead for my intended projects :)