khoih-prog/ESP32TimerInterrupt #12: QUESTION: Is it ok to control i2c device using timer interrupt?

albertlt, 2021-04-09:
Hello. This is a question, not a bug. I have an i2c relay component in my setup. I need the relay to be fired at exact interval without being interrupted. I am aware that anything executed by interrupt must be fast in nature. Is sending i2c command to turn on/off relay acceptable to be executed inside interrupt? I've also read warning about possible data loss in serial communication. While I don't use serial, I do use wifi. Is there any possibility to have data loss if I use interrupt to control i2c device?

khoih-prog, 2021-04-09:
Thanks for your interests in the library.

**It's not advisable and don't use that kind of slow i2c code inside ISRs.**
You can set a **flag** inside ISR, then use a **fast** software timer (every 10-100ms, for example) to trigger the i2c code.

I'm sorry I won't spend time to deal with this issue not relating to the library. It's better you ask for help and questions on [**Arduino forum**](https://forum.arduino.cc/index.php?board=4.0)

Good Luck,

closed 2021-04-09 by khoih-prog

albertlt, 2021-04-09:
that is enough answer for me. thank you :)