khoih-prog/ESP32TimerInterrupt #16: Wrong delay by ISR Timer

sacculus, 2021-07-15:
Hi
I tried to use ISR_Timer to limit duty cycle of battery powered device but timer fired not so as requested.
I have set delay 30 secs but timer callback executed at 10 seconds, which about 3 times afore.
Tell me pls where is my mistake.

My code:
****************************************************
`#define mS_TO_S_FACTOR             1000
#define uS_TO_mS_FACTOR            1000
#define uS_TO_S_FACTOR          1000000

#define HW_TIMER_INTERVAL_MS         50

#include "ESP32TimerInterrupt.h"
#include "ESP32_ISR_Timer.h"

ESP32Timer HWTimer(0);
ESP32_ISR_Timer ISR_Timer;

void IRAM_ATTR handleHWTimer(void) {
  ISR_Timer.run();
}

void IRAM_ATTR handleTimer() {
  Serial.println("Job time exceeded");
  goToSleep(p_job_retry_interval_sec.value());
}

// This function called in setup()
void setupSystem() {
  Serial.println("Setting up system");

  Serial.print(F("CPU Frequency = ")); Serial.print(F_CPU / 1000000); Serial.println(F(" MHz"));
...
  // Starting up timers
  HWTimer.attachInterruptInterval(HW_TIMER_INTERVAL_MS * uS_TO_mS_FACTOR, handleHWTimer);
  int timer = ISR_Timer.setInterval((uint32_t)p_job_timeout_sec.value() * mS_TO_S_FACTOR, handleTimer);
  ISR_Timer.enable(timer);
...
}`
****************************************************
Where, routine "p_job_timeout_sec.value()" returns value 30.

Serial console log:
****************************************************
12:07:46.696 -> Setting up system
12:07:46.696 -> CPU Frequency = 240 MHz
...
12:07:57.679 -> Job time exceeded
12:07:57.679 -> System going to sleep for 3600 seconds
****************************************************

khoih-prog, 2021-07-15:
Hi,

I think there is **something wrong with other part of your code**. Possibly a conflict in using **Timer0** of ESP32. Try to change to **Timer1**, etc. by

```
ESP32Timer HWTimer(1);
```

I'm testing using this code and OK

```
#define mS_TO_S_FACTOR 1000
#define uS_TO_mS_FACTOR 1000
#define uS_TO_S_FACTOR 1000000

#define HW_TIMER_INTERVAL_MS 50

#include "ESP32TimerInterrupt.h"
#include "ESP32_ISR_Timer.h"

ESP32Timer HWTimer(0);
ESP32_ISR_Timer ISR_Timer;

#define p_job_retry_interval_sec_value    30

void goToSleep()
{
  Serial.println("goToSleep");
}

void IRAM_ATTR handleHWTimer()
{
  ISR_Timer.run();
}

void IRAM_ATTR handleTimer() 
{
  Serial.println("Job time exceeded");
  goToSleep();
}

// This function called in setup()
void setup() 
{
  Serial.begin(115200);
  while (!Serial);

  delay(100);
  
  Serial.println("Setting up system");

  Serial.print(F("CPU Frequency = ")); Serial.print(F_CPU / 1000000); Serial.println(F(" MHz"));

  // Starting up timers
  HWTimer.attachInterruptInterval(HW_TIMER_INTERVAL_MS * uS_TO_mS_FACTOR, handleHWTimer);
  int timer = ISR_Timer.setInterval(p_job_retry_interval_sec_value * mS_TO_S_FACTOR, handleTimer);

  ISR_Timer.enable(timer);
}

void loop()
{
  
}
```

---

### Terminal Output

```
14:45:47.918 -> Setting up system
14:45:47.918 -> CPU Frequency = 240 MHz
14:46:17.913 -> Job time exceeded
14:46:17.913 -> goToSleep
14:46:47.923 -> Job time exceeded
14:46:47.923 -> goToSleep
14:47:17.925 -> Job time exceeded
14:47:17.925 -> goToSleep
14:47:47.899 -> Job time exceeded
14:47:47.899 -> goToSleep
14:48:17.919 -> Job time exceeded
14:48:17.919 -> goToSleep
```

I'm not going to spend time here and only you can find out and fix your issue.

closed 2021-07-15 by khoih-prog

sacculus, 2021-07-15:
OK
Thank you! I'll try

sacculus, 2021-07-16:
Hi
I've tried all four timers, tried to change delay to 90 seconds, results were same for all - delay is about 11 seconds...
I'll very grateful if you'd help me to investigate source of problem - how to debug/trace timer settings, check prescaler etc? It is really strange behaviour, becuse I've used ESP8266TimerInterrupt library without any problem.
I enabled defines for debug output of library, this is output

`09:16:47.620 -> CPU Frequency = 240 MHz
09:16:47.620 -> Enable sensors power... done
09:16:47.620 -> [TISR] ESP32TimerInterrupt: _timerNo = 3 , _fre = 1000000.00
09:16:47.620 -> [TISR] _count = 0 - 50000
`
Thank you a lot!

khoih-prog, 2021-07-16:
Sorry I don't have time to spend on solving your problem as it has nothing to do with the library.
I suggest you start from the working example, then moving forward gradually.

Good Luck,

sacculus, 2021-07-16:
Hi
I've slightly modified library to write debug to serial in ESP32_ISR_Timer::run() and found that timer was initialized with wrong value. So that is full my mistake.
Sorry that I stealed your time.