khoih-prog/ESP32TimerInterrupt #17: Deprecation warnings with esp32-v2.0.0-rc1

marcboon, 2021-08-14:
Arduino IDE version: 1.8.15
ESP32 Core Version 2.0.0-rc1
OS: Windows 10

Using ESP32TimerInterrupt with **esp32-v2.0.0-rc1** I get deprecation warnings:

``
ESP32_S2_TimerInterrupt.h:186:33: warning: 'esp_err_t timer_spinlock_take(timer_group_t)' is deprecated [-Wdeprecated-declarations]
``

and

``
ESP32_S2_TimerInterrupt.h:202:33: warning: 'esp_err_t timer_spinlock_give(timer_group_t)' is deprecated [-Wdeprecated-declarations]
``

Code runs as expected, but for how long?

khoih-prog, 2021-08-15:
Hi,

Thanks for your interest in the library and your remind of the **deprecated** functions in new **ESP32 core v2.0.0-rc1+**

As you already knew, the new ESP32 core is still in its **infancy Pre-Release stage**, with many more unexpected possible **breaking** changes. That's why any new development to support the new breaking core still has to wait and is not finalized yet.

Anyway, I've just finished coding the new unified **ESP32_New_TimerInterrupt** library, which is still in the final testing stage, and will be released within next week.

The new library has to be created to unify the 3 different ones, with the breaking changes of the new ESP32 core **v2.0.0+**, and possibly can't be backward compatible to core **v1.0.6-**

Then the  **ESP32_New_ISR_Servo** library, based on that new  **ESP32_New_TimerInterrupt**  library will be also released.

Be patient,

khoih-prog, 2021-08-15:
The new unified [**ESP32_New_TimerInterrupt** Library](https://github.com/khoih-prog/ESP32_New_TimerInterrupt), for all ESP32, ESP32_S2 and ESP32_C3 has just been published.

No more deprecation warnings from new [**esp32 core v2.0.0-rc1**](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-rc1), hopefully from future cores as well.

Please test and report any new issue.

closed 2021-08-15 by khoih-prog

khoih-prog, 2021-08-16:
FYI

The new unified [**ESP32_New_ISR_Servo** Library](https://github.com/khoih-prog/ESP32_New_ISR_Servo), for all ESP32, ESP32_S2 and ESP32_C3 has just been published.

No more deprecation warnings from new [**esp32 core v2.0.0-rc1**](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-rc1), hopefully from future cores as well.

---

### Releases v1.0.0

1. Basic 16 ISR-based servo controllers using 1 hardware timer for ESP32, ESP32_S2, ESP32_C3-based board
2. Tested with [ESP32 core v2.0.0-rc1+](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-rc1)