khoih-prog/ESP32TimerInterrupt #22: Am I missing a library include ??

stevenmowen, 2022-03-25:
Using Platformio, arduino framework and esp32 dev board, when compiling the example RPM_Measure I get an error message: "pio\libdeps\esp32dev\ESP32TimerInterrupt\src/ESP32TimerInterrupt.hpp:372:66: error: 'TIMER_INTR_T0' was not declared in this scope" along with several other missing definitions.

Do I require some other library besides the ESP32TimerInterrupt ??

stevenmowen, 2022-03-25:
Moving this to discussions.

closed 2022-03-25 by stevenmowen

khoih-prog, 2022-03-25:
As specified in [**Prerequisites**](https://github.com/khoih-prog/ESP32TimerInterrupt#prerequisites), the library requires ESP32 core v2.0.2, which PIO doesn't officially support yet.

You can follow [**Imported Arduino IDE project does not build** #10](https://community.platformio.org/t/imported-arduino-ide-project-does-not-build/26700/10) and use in `platformio.ini`

```
platform = https://github.com/Jason2866/platform-espressif32.git#52fb773
```

Research many more threads in PIO mentioning about the same ESP32 core v2.0.2 issue.