khoih-prog/ESP32TimerInterrupt #23: TIMER_INTR_0 Not defined

robkwhitehouse, 2022-05-09:
Hi,
Thankyou for publishing your ESP32TimerInterrupt library. It appears to offer exactly what I need but I am having a little trouble compiling it. I get compiler errors saying that a couple of macros are not defined. They are TIMER_INTR_0 and TIMER_INTR_1.

I am using the following build environment;
Arduino IDE V 1.8.19 on Ubuntu
Board: ESP32 Devkit 1

I tried grepping around in the libraries folder for these macros but couldn't  find them. Am I missing an include file?

Thanks for your help.

Bob W

robkwhitehouse, 2022-05-09:
I should add that the file that is failing to compile is ESP32TimerInterrupt.hpp

khoih-prog, 2022-05-09:
If you have an issue, first, have a look at the 

1. Instructions, e.g. [**Prerequisites**](https://github.com/khoih-prog/ESP32TimerInterrupt#prerequisites)
2. closed issues, e.g. [**Am I missing a library include ??** #22](https://github.com/khoih-prog/ESP32TimerInterrupt/issues/22)

then you'll know you have to upgrade the **ESP32 core to v2.0.2 or v2.0.3**

closed 2022-05-09 by khoih-prog